COVESA/vsomeip #805: why E2E Profile 4 and  Profile 5  can only be used for instances [1-255]"

GHARBIAchref, 2024-11-11:
### vSomeip Version

v3.4.10

### Boost Version

1.83

### Environment

windows 10

### Describe the bug

 why E2E Profile 4 and  Profile 5  can only be used for instances ### [1-255]"
  VSOMEIP_ERROR << "E2E Profile 4 can only be used for instances [1-255]";

duartenfonseca, 2024-11-11:
do you have reproduction steps for the error?

GHARBIAchref, 2024-11-11:
[image]
When I try to apply the protection mechanism of profile 04 to a service that has an instance >255 , this error message appears, and the protection  Profile 04 does not apply.

duartenfonseca, 2024-11-12:
do you have an instance value bigger than 255?
[image]
vsomeip-lib/implementation/e2e_protection/src/e2e/profile/profile04/checker.cpp::24

GHARBIAchref, 2024-11-12:
@duartenfonseca   
Hello ,
  es, my application has an instance value greater than 255. Is it possible to apply the E2E P04 fetcher in this application?

duartenfonseca, 2024-11-12:
i don't think it is possible. this was the way it was implemented. maybe you can try to use profile 7 or profile 1 which do not have that check.

duartenfonseca, 2024-11-18:
this is implemented like this by design, it does not allow a bigger instance value than 255

closed 2024-11-18 by duartenfonseca