COVESA/vsomeip #809: [BUG]: <Compliance issue that offer the same serviceID-instanceID with different major versions is not supported>

zhaoxin39913, 2024-11-15:
### vSomeip Version

v3.4.10

### Boost Version

1.7.6

### Environment

Ubuntu 20.04

### Describe the bug

Hi vsomeip experts,

To follow up https://github.com/COVESA/vsomeip/issues/797. I have tested PRS_SOMEIPSD_00512 in SOME/IP SD Spec and confirmed that offering the same serviceID-instanceID with different major versions within the same VLAN is not supported in current vsomeip library.

In my test, I have set up two devices running different vsomeip applications.  device A offers service 0x1234/0x5678 with majvor version 0xA, deviceB offers service 0x1234/0x5678 with major version 0xB.  After device A has offered the service, next when device B offers the same service with version 0xB, it always reports below error message:

` [error] routing_manager_impl::handle_local_offer_service: rejecting service registration. Application: 0201 is trying to offer [1234.5678.11.0] already offered ..`

And from the wireshark I confirmed that the service 0x1234/0x5678/0xB is not offered.  It seems like a missing feature that vsomeip stack need to support  PRS_SOMEIPSD_00512 (Supporting multiple versions of the same service),

### Reproduction Steps

100% reproduced.

### Expected behaviour

vsomeip stack need to support  PRS_SOMEIPSD_00512 (Supporting multiple versions of the same service)

### Logs and Screenshots

[error] routing_manager_impl::handle_local_offer_service: rejecting service registration. Application: 0201 is trying to offer [1234.5678.11.0] already offered

cpvep, 2024-11-15:
I have the same issue. Can somebody please help with this issue

cpvep, 2024-11-15:
I encountered the same issue. Can someone please help with this?

KareemLMR, 2024-11-16:
Hello,
How did you set the major version of the services you offered?
Have you set the major version in offer_service API:
[image]

KareemLMR, 2024-11-23:
Hello @zhaoxin39913 
I need to check something. Open the vsomeip source code and navigate to the file implementation/routing/src/routing_manager_base.cpp at this function routing_manager_base::offer_service to be specific
[image]

And add a log to this if condition marked by the red arrow: 
if (!its_info->is_local()) {
            return false;
        }
and if you can add more logs throughout this function then recompile and reinstall the binaries and let me see the logs of your test. If the log you added in that if condition appeared, then there is an issue as you said. If not then it is something else we will see.

zhaoxin39913, 2024-11-28:
Hi @KareemLMR,

I have made below changes per as your request:

    if (!its_info->is_local()) {
        VSOMEIP_INFO << "rm_base::offer_service !its_info->is_local ("
                << std::hex << std::setfill('0')
                << std::setw(4) << _client << "): ["
                << std::setw(4) << _service << "."
                << std::setw(4) << _instance << ":"
                << std::dec
                << static_cast<std::uint32_t>(its_info->get_major()) << ":"
                << its_info->get_minor() << "] passed: "
                << static_cast<std::uint32_t>(_major) << ":"
                << _minor;
        return false;
And after change I have triggered "notify-sample" to offer the second service instance 0x1234/0x5678 with version 0xA (There is already a service instance 0x1234/0x5678 with version 0x12 offered by another device within the same network). And from the wireshark we can see that the first time notify-sample was successful to offer the service 0x1234/0x5678 version 0xA, but after call stop_offer() and then start_offer() in "notify-sample", it failed to offer the service again. Below are full log messages from notify-sample:

```
zhax@zhax-desktop:~/work/COVESA/src/vsomeip-3.4.10/build/examples$ VSOMEIP_APPLICATION_NAME=service-sample ./notify-sample
2024-11-28 14:44:42.707338 [info] Using configuration folder: "/home/zhax/work/COVESA/src/config".
2024-11-28 14:44:42.709307 [info] Parsed vsomeip configuration in 1ms
2024-11-28 14:44:42.709649 [info] Configuration module loaded.
2024-11-28 14:44:42.709911 [info] Security disabled!
2024-11-28 14:44:42.710142 [info] Initializing vsomeip (3.4.10) application "service-sample".
2024-11-28 14:44:42.711782 [info] Instantiating routing manager [Host].
2024-11-28 14:44:42.714633 [info] create_routing_root: Routing root @ /tmp/vsomeip-0
2024-11-28 14:44:42.716890 [info] Service Discovery enabled. Trying to load module.
2024-11-28 14:44:42.725771 [info] Service Discovery module loaded.
2024-11-28 14:44:42.726925 [info] Application(service-sample, 1277) is initialized (11, 100).
2024-11-28 14:44:42.727391 [info] offer_event: Event [1234.5678.9999] uses configured cycle time 0ms
2024-11-28 14:44:42.728266 [info] REGISTER EVENT(1277): [1234.5678.9999:is_provider=true]
2024-11-28 14:44:42.728753 [info] Starting vsomeip application "service-sample" (1277) using 2 threads I/O nice 255
2024-11-28 14:44:42.731711 [info] main dispatch thread id from application: 1277 (service-sample) is: ffffbc79f060 TID: 189214
2024-11-28 14:44:42.732672 [info] shutdown thread id from application: 1277 (service-sample) is: ffffbbf8f060 TID: 189215
2024-11-28 14:44:42.733317 [info] create_local_server: Listening @ /tmp/vsomeip-1277
2024-11-28 14:44:42.734978 [info] OFFER(1277): [1234.5678:10.0] (true)
2024-11-28 14:44:42.732532 [info] Client [1277] routes unicast:192.168.225.3, netmask:255.255.255.0
Setting event (Length=1).
2024-11-28 14:44:42.739600 [info] Watchdog is disabled!
Application service-sample is registered.
2024-11-28 14:44:42.741732 [info] io thread id from application: 1277 (service-sample) is: ffffbec44020 TID: 189210
2024-11-28 14:44:42.741929 [info] io thread id from application: 1277 (service-sample) is: ffffbaf6f060 TID: 189217
2024-11-28 14:44:42.743662 [info] vSomeIP 3.4.10 | (default)
2024-11-28 14:44:42.744930 [info] Network interface "eth0" state changed: up
2024-11-28 14:44:42.746798 [info] Route "default route (0.0.0.0/0) if: eth0 gw: 192.168.225.1" state changed: up
2024-11-28 14:44:42.748222 [info] udp_server_endpoint_impl: SO_RCVBUF is: 212992 (1703936) local port:30490
2024-11-28 14:44:42.748774 [debug] Joining to multicast group 224.0.0.1 from 192.168.225.3
2024-11-28 14:44:42.750352 [info] udp_server_endpoint_impl: SO_RCVBUF is: 212992 (1703936) local port:30509
2024-11-28 14:44:42.750775 [info] SOME/IP routing ready.
2024-11-28 14:44:42.750875 [info] udp_server_endpoint_impl<multicast>: SO_RCVBUF is: 212992 (1703936) local port:30490
2024-11-28 14:44:42.752166 [warning] Route "224.0.0.1/32 if: eth0 gw: n/a" state changed: up
2024-11-28 14:44:43.334512 [error] routing_manager_impl::add_routing_info: rejecting routing info. Remote: 192.168.225.2 is trying to offer [1234.5678.18.878082192] on port 12344 offered previously on this node: [1234.5678.10.0]
Setting event (Length=2).
Setting event (Length=3).
2024-11-28 14:44:45.339291 [error] routing_manager_impl::add_routing_info: rejecting routing info. Remote: 192.168.225.2 is trying to offer [1234.5678.18.878082192] on port 12344 offered previously on this node: [1234.5678.10.0]
Setting event (Length=4).
Setting event (Length=5).
2024-11-28 14:44:47.349538 [error] routing_manager_impl::add_routing_info: rejecting routing info. Remote: 192.168.225.2 is trying to offer [1234.5678.18.878082192] on port 12344 offered previously on this node: [1234.5678.10.0]
Setting event (Length=6).
Setting event (Length=7).
2024-11-28 14:44:49.357992 [error] routing_manager_impl::add_routing_info: rejecting routing info. Remote: 192.168.225.2 is trying to offer [1234.5678.18.878082192] on port 12344 offered previously on this node: [1234.5678.10.0]
Setting event (Length=8).
Setting event (Length=9).
2024-11-28 14:44:51.367642 [error] routing_manager_impl::add_routing_info: rejecting routing info. Remote: 192.168.225.2 is trying to offer [1234.5678.18.878082192] on port 12344 offered previously on this node: [1234.5678.10.0]
Setting event (Length=1).
[... cut ...]
2024-11-28 14:45:42.762672 [info] vSomeIP 3.4.10 | (default)
2024-11-28 14:45:42.825814 [info] rm_base::offer_service !its_info->is_local (1277): [1234.5678:18:878082192] passed: 10:0
2024-11-28 14:45:42.831568 [error] routing_manager_impl::handle_local_offer_service: rejecting service registration. Application: 1277 is trying to offer [1234.5678.10.0]] already offered remotely
2024-11-28 14:45:42.832564 [info] offer_service (1277): [1234.5678:10.0] (true) not offering, returned from handle_local_o
```

KareemLMR, 2024-11-28:
Ok, I see the log you added appeared. That's good. I saw what I wanted to see. Let me try a change at my side and send it to you to try when it is successful.

zhaoxin39913, 2024-11-29:
Thanks @KareemLMR.  I think here we should allow to offer the second same service instance if its major verion is different from the existing same service instance in the network.

KareemLMR, 2024-12-16:
Hello @zhaoxin39913 I thought of something that might make the change much simpler instead of changing the structure but I need to check it is working with you before going on. Can you please test after applying this patch?

With this patch you should be able to offer multiple services instances with different major versions and request any of them.
[PRS_SOMEIPSD_00512.patch](https://github.com/user-attachments/files/18157873/PRS_SOMEIPSD_00512.patch)

But this patch is based on tag: 3.5.3 so you have to use that version for the patch to apply.

KareemLMR, 2024-12-20:
You can try this change. I was able to offer 2 services with only different major versions through it but I need you to verify it. Here is the code examples:
service1.cpp:

#include <vsomeip/vsomeip.hpp>

#define SAMPLE_SERVICE_ID 0x1234
#define SAMPLE_INSTANCE_ID 0x5678

std::shared_ptr< vsomeip::application > app;

int main() {
    vsomeip_v3::unique_version_t unique = vsomeip_v3::get_unique_version(SAMPLE_INSTANCE_ID, 10);
    app = vsomeip::runtime::get()->create_application("Hello");
    app->init();
    app->offer_service(SAMPLE_SERVICE_ID, unique, 10, 0);
    app->start();
}

service2.cpp:

#include <vsomeip/vsomeip.hpp>

#define SAMPLE_SERVICE_ID 0x1234
#define SAMPLE_INSTANCE_ID 0x5678

std::shared_ptr< vsomeip::application > app;

int main() {
    vsomeip_v3::unique_version_t unique = vsomeip_v3::get_unique_version(SAMPLE_INSTANCE_ID, 11);
    app = vsomeip::runtime::get()->create_application("World");
    app->init();
    app->offer_service(SAMPLE_SERVICE_ID, unique, 11, 0);
    app->start();
}

Also for the json file, I supported adding "major" : "value" just as instance in your service definition so don't forget that.
Please tell me the results of your testing.

zhaoxin39913, 2024-12-26:
Thanks @KareemLMR , I will test your patch and let you know the result.

zhaoxin39913, 2024-12-31:
Hi @KareemLMR ,

Thanks for sharing the sample code. One question here, for client API request_service(), Do we still support passing ANY_MAJOR when build the unique_version_t parameter? Or we must pass a non-0xFF version ?

virtual void request_service(service_t _service, unique_version_t _unique,
            major_version_t _major = ANY_MAJOR,
            minor_version_t _minor = ANY_MINOR,
            bool _use_exclusive_proxy = false) = 0;

KareemLMR, 2024-12-31:
Hello @zhaoxin39913 
Unfortunately the current patch set in case of using unique version to use the new feature of offering multiple instances with different major versions you have to request the service specifically by its major version
Using 0xFF as major won't work as expected in the current version but I was planning to fix that in another patch set just after checking your test results to know if it is the only thing missing before committing another change.
Have you tested it with requesting the specific major version and the rest of APIs?

zhaoxin39913, 2025-01-02:
Thanks @KareemLMR , I will update my code accordingly to test the patch.

closed 2025-11-19 by fcmonteiro