COVESA/vsomeip #810: [BUG]: <title>

Lynn-1997, 2024-11-16:
### vSomeip Version

V3.4.10

### Boost Version

1.76

### Environment

QNX7.1

### Describe the bug

Start in this order occur erro:
1、start routingmanagerd, as routing
2、start notify-sample, offer service 
3、start subscribe-sample, subscribe service 
subscribe-sample can not receive SUBSCRIBE ACK , can not receive message.

Start in this order:
1、start notify-sample, as routing and offer service 
2、start subscribe-sample, subscribe service 
subscribe-sample can receive SUBSCRIBE ACK , and can receive message.

### Reproduction Steps

**#1 Compile V3.4.10 demo based on source code.**
**#2 Execute within QNX system step**
1、start routingmanagerd, as routing
env VSOMEIP_CONFIGURATION=/mnt/etc/vsomeip.json; VSOMEIP_APPLICATION_NAME=routingmanagerd; ./bin/routingmanagerd>/Log_QNX/vsomeip_log/routingmanagerd.log&
2、start notify-sample, offer service 
env VSOMEIP_CONFIGURATION=/mnt/etc/vsomeip.json; VSOMEIP_APPLICATION_NAME=notify-sample; ./bin/notify-sample>/Log_QNX/vsomeip_log/notify.log&
3、start subscribe-sample, subscribe service 
env VSOMEIP_CONFIGURATION=/mnt/etc/vsomeip.json; VSOMEIP_APPLICATION_NAME=subscribe-sample; ./bin/subscribe-sample>/Log_QNX/vsomeip_log/subscribe.log&
config:
[vsomeip.json](https://github.com/user-attachments/files/17784096/vsomeip.json)

### Expected behaviour

1、subscribe-sample log include:
SUBSCRIBE ACK(6666): [1234.5678.4465.ffff]
2、subscribe-sample log can recieve notify message.

### Logs and Screenshots

[notify.log](https://github.com/user-attachments/files/17784084/notify.log)
[routingmanagerd.log](https://github.com/user-attachments/files/17784085/routingmanagerd.log)
[subscribe.log](https://github.com/user-attachments/files/17784086/subscribe.log)

joeyoravec, 2024-12-05:
I'm not too familiar with this example code, but I see it's registering for SAMPLE_EVENT_ID == 0x8778 which we can see in your log:
```
1970-01-01 00:06:24.842109 [info] REGISTER EVENT(1277): [1234.5678.8778:eventtype=2:is_provided=true:reliable=255]
```
however the vsomeip.json that you attached has events 0x0777 ... 0x0779 without the leading 8. I checked the sample config files and the event numbers vary between the local and remote examples. I don't know if this is significant

Lynn-1997, 2024-12-12:
{
    "unicast" : "192.168.1.1",
	"logging": {
        "level": "debug",
        "console": "true",
		"file": {
            "enable": "true",
            "path": " /Log_QNX/vsomeip_log/routingmanagerd.log" 
        },
        "dlt": "false" 
    },
    "applications" :
    [
        {
            "name" : "notify-sample",
            "id" : "0x1277"
        },
        {
            "name" : "subscribe-sample",
            "id" : "0x1344"
        },
        {
            "name" : "routingmanagerd",
            "id" : "0x6666"
        }
    ],
    "services" :
    [
        {
            "service" : "0x1234",
            "instance" : "0x5678",
            "unreliable" : "30509",
            "events" :
            [
                {
                    "event" : "0x8778",
                    "is_field" : "true",
                    "is_reliable" : "false"
                }
            ],
            "eventgroups" :
            [
                {
                    "eventgroup" : "0x4465",
                    "events" : [ "0x8778" ]
                }
            ]
        }
    ],
    "routing" : "routingmanagerd",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "224.244.224.245",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "3",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }
}

Lynn-1997, 2024-12-12:
**notify-sample log:**
1970-01-01 00:00:44.444185 [info] Using configuration file: "/mnt/etc/vsomeip.json".
1970-01-01 00:00:44.444187 [info] Parsed vsomeip configuration in 7ms
1970-01-01 00:00:44.444187 [info] Configuration module loaded.
1970-01-01 00:00:44.444187 [info] Security disabled!
1970-01-01 00:00:44.444187 [info] Initializing vsomeip (3.4.10) application "notify-sample".
1970-01-01 00:00:44.444187 [info] Instantiating routing manager [Proxy].
1970-01-01 00:00:44.444187 [info] Client [1277] is connecting to [0] at /var/vsomeip/vsomeip-0
1970-01-01 00:00:44.444187 [info] Application(notify-sample, 1277) is initialized (11, 100).
1970-01-01 00:00:44.444187 [info] offer_event: Event [1234.5678.8778] uses configured cycle time 0ms
1970-01-01 00:00:44.444187 [info] Starting vsomeip application "notify-sample" (1277) using 2 threads I/O nice 255
Setting event (Length=1).
1970-01-01 00:00:44.444187 [info] main dispatch thread id from application: 1277 (notify-sample) is: 4
1970-01-01 00:00:44.444187 [info] ## test ## routing_manager_client::start
1970-01-01 00:00:44.444187 [info] shutdown thread id from application: 1277 (notify-sample) is: 5
1970-01-01 00:00:44.444197 [info] io thread id from application: 1277 (notify-sample) is: 6
1970-01-01 00:00:44.444200 [info] io thread id from application: 1277 (notify-sample) is: 1
1970-01-01 00:00:44.444207 [info] rmp::on_message<1277>: 01 00 00 00 00 02 00 00 00 77 12 
1970-01-01 00:00:44.444207 [info] ## test ## disable VSOMEIP_ENABLE_COMPAT
1970-01-01 00:00:44.444207 [info] ##test 222## received protocol
1970-01-01 00:00:44.444207 [info] ##test 111## received protocol
1970-01-01 00:00:44.444207 [info] ##test## received a message with command 1
1970-01-01 00:00:44.444212 [info] create_local_server: Listening @ /var/vsomeip/vsomeip-1277
1970-01-01 00:00:44.444212 [info] Client 1277 (notify-sample) successfully connected to routing  ~> registering..
1970-01-01 00:00:44.444217 [info] Registering to routing manager @ vsomeip-0
1970-01-01 00:00:44.444234 [info] rmp::on_message<1277>: 05 00 00 66 66 07 00 00 00 00 02 00 00 00 77 12 
1970-01-01 00:00:44.444236 [info] ## test ## disable VSOMEIP_ENABLE_COMPAT
1970-01-01 00:00:44.444239 [info] ##test 222## received protocol
1970-01-01 00:00:44.444239 [info] ##test 111## received protocol
1970-01-01 00:00:44.444239 [info] ##test## received a message with command 5
1970-01-01 00:00:44.444239 [info] Application/Client 1277 (notify-sample) is registered.
Application notify-sample is registered.
Setting event (Length=2).
1970-01-01 00:00:45.454198 [info]  #test 887 : routing_manager_client::on_message: 
1970-01-01 00:00:45.454199 [info]  #test 931 : routing_manager_client::on_message: 
Setting event (Length=3).
1970-01-01 00:00:46.464217 [info]  #test 887 : routing_manager_client::on_message: 
1970-01-01 00:00:46.464217 [info]  #test 931 : routing_manager_client::on_message: 
Setting event (Length=4).
1970-01-01 00:00:47.474228 [info]  #test 887 : routing_manager_client::on_message: 
1970-01-01 00:00:47.474228 [info]  #test 931 : routing_manager_client::on_message: 
Setting event (Length=5).
1970-01-01 00:00:48.484241 [info]  #test 887 : routing_manager_client::on_message: 
1970-01-01 00:00:48.484242 [info]  #test 931 : routing_manager_client::on_message: 
Setting event (Length=6).
1970-01-01 00:00:49.494253 [info]  #test 887 : routing_manager_client::on_message: 
1970-01-01 00:00:49.494255 [info]  #test 931 : routing_manager_client::on_message: 
Setting event (Length=7).
1970-01-01 00:00:50.504268 [info]  #test 887 : routing_manager_client::on_message: 
1970-01-01 00:00:50.504268 [info]  #test 931 : routing_manager_client::on_message: 
Setting event (Length=8).
1970-01-01 00:00:51.514282 [info]  #test 887 : routing_manager_client::on_message: 
1970-01-01 00:00:51.514283 [info]  #test 931 : routing_manager_client::on_message: 
Setting event (Length=9).
1970-01-01 00:00:52.524298 [info]  #test 887 : routing_manager_client::on_message: 
1970-01-01 00:00:52.524298 [info]  #test 931 : routing_manager_client::on_message: 
Setting event (Length=1).
1970-01-01 00:00:53.534318 [info]  #test 887 : routing_manager_client::on_message: 
1970-01-01 00:00:53.534322 [info]  #test 931 : routing_manager_client::on_message: 
Setting event (Length=2).
1970-01-01 00:01:04.644420 [info]  #test 887 : routing_manager_client::on_message: 
1970-01-01 00:01:04.644423 [info]  #test 931 : routing_manager_client::on_message: 

1970-01-01 00:01:04.644429 [info] rmp::on_message<1277>: 05 00 00 66 66 07 00 00 00 00 02 00 00 00 44 13 
1970-01-01 00:01:04.644434 [info] ## test ## disable VSOMEIP_ENABLE_COMPAT
1970-01-01 00:01:04.644434 [info] ##test 222## received protocol
1970-01-01 00:01:04.644439 [info] ##test 111## received protocol
1970-01-01 00:01:04.644440 [info] ##test## received a message with command 5
1970-01-01 00:01:04.644448 [info] rmp::on_message<1277>: 12 00 00 44 13 0b 00 00 00 34 12 78 56 65 44 00 ff ff 00 00 
1970-01-01 00:01:04.644448 [info] ## test ## disable VSOMEIP_ENABLE_COMPAT
1970-01-01 00:01:04.644448 [info] ##test 222## received protocol
1970-01-01 00:01:04.644448 [info] ##test 111## received protocol
1970-01-01 00:01:04.644448 [info] ##test## received a message with command 18
1970-01-01 00:01:04.644448 [info] SUBSCRIBE(1344): [1234.5678.4465:ffff:0]

Setting event (Length=3).
1970-01-01 00:01:05.654442 [info]  #test 887 : routing_manager_client::on_message: 
1970-01-01 00:01:05.654442 [info]  #test 931 : routing_manager_client::on_message: 
Setting event (Length=4).

Lynn-1997, 2024-12-12:
I checked the configuration and found that there was indeed a problem with the configuration. I updated the configuration and verified again, but subscribe-sample  could not receive SUBSCRIBE ACK.

>  @[joeyoravec](/joeyoravec)

rangasridhar07-droid, 2026-02-12:
@Lynn-1997  Should we run all three application simultaneously?Routingmanager,notify-sample,subscribe-sample.
Can you check if you are able to capture wireshark packets for Service discovery?
Does all three apps stay on same ip and on same target?