COVESA/vsomeip #816: [BUG]: Honor TMPDIR environment variable if defined for Linux systems

srchaitu2008, 2024-11-25:
### vSomeip Version

v3.4.10

### Boost Version

1.81

### Environment

Debian 11

### Describe the bug

​​Honor TMPDIR environment variable if defined for Linux systems to allow the vsomeip.lck and other temp files to be written in the path defined in TMPDIR env variable instead of default /tmp folder​.

KareemLMR, 2024-11-26:
Hello @srchaitu2008 
I think it is already implemented here:
[image]
You should just pass the TMPDIR you want to the cmake command as cmake -DBASE_PATH=<YOUR BASE PATH> ..

I found it also mentioned in the user guide here:
[image]

closed 2024-12-12 by duartenfonseca

srchaitu2008, 2024-12-12:
Hi @KareemLMR ,

I agree with you. Whatever you have mentioned( cmake -DBASE_PATH= ) is **compile time** specific. The solution which I have submitted is **runtime specific** (i.e.) at **runtime** the **TMPDIR** environment variable will be checked and used if available, else the default path would be considered. 

I hope this clarifies your concern.

Thanks,
Ravi

duartenfonseca, 2024-12-12:
@srchaitu2008 what is the use case where you have to do it like this? wouldn't it be enough being able to set this path on compile time?