COVESA/vsomeip #825: [BUG]: <title>The server closing the TCP connection when stopping all services.

AGN-2022, 2024-12-17:
### vSomeip Version

v3.4.10

### Boost Version

1.59

### Environment

Ubuntu 20.04

### Describe the bug

1. Server Offering services.
2. Client initiates TCP connection by sending [SYN]. [by 3-way handshake connection established]
3. Client subscribes to a particular Event group offered by Server.
4. Server sends ACK to client.
5. Client calls method "SuspendInterface" to Server. [Intended to trigger Stop Offer from Server]
6. Server sends Stop Offer Service Entry.
7. Server initiates TCP connection close request by sending [FIN, ACK] to Client.

The TCP connection close request should not get triggered by Server, in this case this is the issue from VSOME/IP stack.

I observed the same behavior in 3.1.20.3 and 3.4.10.
[TCP_close_issue_pcap_RM_logs.zip](https://github.com/user-attachments/files/18164271/TCP_close_issue_pcap_RM_logs.zip)

zhaoxin39913, 2025-01-22:
Can you point out which requirement in the SOME/IP Spec defines such behaivor ?

AGN-2022, 2025-01-23:
> Can you point out which requirement in the SOME/IP Spec defines such behaivor ?

Dear @zhaoxin39913 , please check the below Spec.

**[PRS_SOMEIP_00709]**
  Upstream requirements: RS_SOMEIP_00010
The TCP connection shall be closed by the client, when the TCP connection is not
required anymore.

**[PRS_SOMEIP_00710]**
   Upstream requirements: RS_SOMEIP_00010
The TCP connection shall be closed by the client, when all Services using the TCP
connections are not available anymore (stopped or timed out).

**[PRS_SOMEIP_00711]**
Upstream requirements: RS_SOMEIP_00010
The server shall not stop the TCP connection when stopping all services. Give the
client enough time to process the control data to shutdown the TCP connection itself.

Rational:
When the server closes the TCP connection before the client recognized that the TCP
is not needed anymore, the client will try to reestablish the TCP connection.

zhaoxin39913, 2025-01-23:
Hi @AGN-2022 ,

So the expected behavior is the server only closes the listen socket but keeps established connection sockets after stopping the service, right ?

Do you need this only for TC8 certificate ? To my understanding there is no function impact from application PoV. And the Spec also does not define how the server reacts subsequent requests from client after stopping the service but the connection socket is not closed by client.

AGN-2022, 2025-01-24:
Hi @zhaoxin39913 ,

If you observe pcap log, soon after Frame# 28 (which is Stop-Offer), Frame# 30 is triggering TCP-connection close which is not intended behavior as per Spec.

[image]

I'm not sure about TC8 certificate.

I believe _`tcp_server_endpoint_impl::connection`_ APIs can be analyzed and updated in such a way that specific connection established sockets shouldn't be closed from Server point of view.

zhaoxin39913, 2025-02-05:
Hi @AGN-2022 ,

I understood your point. But if it's not for TC8 certificate, what's the real impact to your client application? Or it's just for compliance testing ?

AGN-2022, 2025-02-06:
Hi @zhaoxin39913,

Yes, it is for compliance testing to satisfy the Spec.

I believe COVESA/VSOME/IP must be adhering to AUTOSAR Spec, so what are your thoughts on this? will you be accepting this as non-compliant and fixing the same?

Mythili21CSR121, 2026-08-31:
Hi @fcmonteiro,

Could you please provide the SW version in which this bug was fixed, along with the commit containing the fix?