COVESA/vsomeip #826: [BUG]: Client Random Port Issue in Notify-Subscribe Example using UDP

ayaha55, 2024-12-19:
### vSomeip Version

v3.4.10 | the latest version

### Boost Version

1.74

### Environment

Ubuntu 22.04 LTS

### Describe the bug

When running the **notify-subscribe** example in the **vsomeip** repository with UDP, the server uses the correct port as configured in the JSON configuration file, but the **client** randomly assigns a port on each execution. In addition, during each SOME/IP SD (Service Discovery) subscription, the client’s **IPv4 endpoint** in the **option** changes the port number, which leads to inconsistencies.

I have tried the port range solution mentioned in [Issue #452](https://github.com/COVESA/vsomeip/issues/452), but it didn’t work in my case.
The client’s UDP port in Wireshark fluctuates between port ranges, such as from [3xxxx] to [5xxxx], which is not the expected behavior.

### Expected behaviour

The client should use a fixed UDP port for communication, as specified in the configuration file, and not randomly change the port or fluctuate within a range.

goncaloalmeida, 2025-03-06:
@ayaha55 please provide logs, configurations and reproduction steps.

fcmonteiro, 2025-11-19:
client ports are always random.
you can try to define a range in the config
https://github.com/COVESA/vsomeip/blob/master/documentation/vsomeipConfiguration.md#clients