COVESA/vsomeip #831: Help: How to add configuration option to service discovery?

robsoncouto, 2025-01-03:
### vSomeip Version

v3.5.3

### Boost Version

1.74

### Environment

Ubuntu 22.04

### Describe the bug

Hello, I have used vsomeip to implement a service server and it is working beautifully except for a detail. That service needs to be offered containing a configuration option as described in section 5.1.2.4.1 "Configuration Option" of SOME/IP Service Discovery Protocol Specification R23-11.

That is, in the array of options in the service discovery messages I have to add additional data in key value pairs.
[image]

I could not find on the examples, info on how to customize service discovery offer messages. Investigating the code and generated doc, I could find in documentation of vsomeip_v3::sd::configuration_option_impl Class that there are methods for adding key value pairs to the configuration, which is what I need to do. 

Still, it is not clear how to call these from the application API. I have looking into the json configuration and did not find info there. Also the puplic API in vsomeip::application class does not seem to have methods to customize service discovery offer. update_service_configuration() and set_configuration() are unrelated.

It's the first time I work with SOME/IP. Am I missing something? 

Any help on this is very appreciated, thanks!

duartenfonseca, 2025-01-27:
can you give a use case of this functionality?

robsoncouto, 2025-01-29:
Hello @duartenfonseca. Thanks for the interest!

I understand this would be used to provide more information about the service being offered. Information which can also be used to validate the service provider.  From the specification:

`The configuration option is used to transport arbitrary configuration strings. This allows to encode additional information like the name of a service or its configuration.`

This is a list of key=value pairs, which can be arbitrary and possibly dynamic. I see that the library enables parsing of the configuration option as a receiver, and then options are part of the message. It does does not serialize it as a sender.

I have looked into the codebase and adding this does not look trivial at all. 
- First, the local endpoint does not seem to support passing the configuration option from application layer to the application running the network routing host without heavy modifications. 
- If we consider only hardcoded configuration options, these could be added as part of the service description in the configuration file. But having random key=value pairs there seems a hack.