COVESA/vsomeip #843: [BUG]: Local routing endpoint creation failed. Client ID: 0000: Protocol not available

AhMmedMahmoud, 2025-02-04:
### vSomeip Version

v3.5.4

### Boost Version

1.81

### Environment

QNX

### Describe the bug

vsomeip.json 
{
    "unicast" : "127.0.0.1",
    "logging" :
    {
        "level" : "debug",
        "console" : "true",
        "file" : { "enable" : "false", "path" : "/var/log/vsomeip.log" },
        "dlt" : "false"
    },
    "applications" :
    [
        {
            "name" : "client-sample",
            "id" : "0x1343"
        },
        {
            "name" : "other-client-sample",
            "id" : "0x1344"
        },
        {
            "name" : "service-sample",
            "id" : "0x1277"
        }
    ],
    "services" :
    [
        {
            "service" : "0x1234",
            "instance" : "0x5678",
            "reliable" : { "port" : "30509", "enable-magic-cookies" : "false" },
            "unreliable" : "31000"
        },
        {
            "service" : "0x1235",
            "instance" : "0x5678",
            "reliable" : { "port" : "30506", "enable-magic-cookies" : false },
            "unreliable" : "31000"
        }
    ],
    "routing" : "server-sample",
    "service-discovery" :
    {
        "enable" : "false",
        "multicast" : "224.244.224.245",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "3",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }
}

Service side
------------------------------
# COMMONAPI_CONFIG=/tmp/commonapi.ini ./BroadcastService 
[CAPI][INFO] Loading configuration file '/tmp/commonapi.ini'
[CAPI][INFO] Using default binding 'someip'
[CAPI][INFO] Using default shared library folder '/usr/local/lib/commonapi'
[CAPI][DEBUG] Loading library for local:commonapi.examples.Broadcast:v1_0:commonapi.examples.Methods stub.
[CAPI][INFO] Loading configuration file /etc//commonapi-someip.ini
[CAPI][DEBUG] Added address mapping: local:commonapi.examples.Broadcast:v1_0:commonapi.examples.Methods <--> [1234.5678(1.0)]
[CAPI][VERBOSE] Registering function for creating "commonapi.examples.Broadcast:v1_0" proxy.
[CAPI][INFO] Registering function for creating "commonapi.examples.Broadcast:v1_0" stub adapter.
[CAPI][DEBUG] Loading interface library "libBroadcast-someip.so" succeeded.
[CAPI][INFO] Registering stub for "local:commonapi.examples.Broadcast:v1_0:commonapi.examples.Methods"
2025-01-23 15:28:39.192061  [info] Parsed vsomeip configuration in 0ms
2025-01-23 15:28:39.192061  [info] Configuration module loaded.
2025-01-23 15:28:39.192061  [info] Security disabled!
2025-01-23 15:28:39.192061  [info] Initializing vsomeip (3.5.4) application "service-sample".
2025-01-23 15:28:39.192071  [info] Instantiating routing manager [Host].
2025-01-23 15:28:39.192071  [info] create_routing_root: Routing root @ /var/vsomeip-0
2025-01-23 15:28:39.192071  [error] Local routing endpoint creation failed. Client ID: 0000: Protocol not available
2025-01-23 15:28:39.192081  [warning] Routing root creating (partially) failed. Please check your configuration.
2025-01-23 15:28:39.192081  [info] Service Discovery enabled. Trying to load module.
2025-01-23 15:28:39.192132  [info] Service Discovery module loaded.
2025-01-23 15:28:39.192132  [info] Application(service-sample, 0100) is initialized (11, 100).
2025-01-23 15:28:39.192142  [info] Starting vsomeip application "service-sample" (0100) using 2 threads I/O nice 0
2025-01-23 15:28:39.192142  [info] offer_event: Event [1234.5678.80f2] uses configured cycle time 0ms
2025-01-23 15:28:39.192142  [info] REGISTER EVENT(0100): [1234.5678.80f2:is_provider=true]
2025-01-23 15:28:39.192142  [info] rmi::offer_service added service: 1234 to pending_sd_offers_.size = 1
2025-01-23 15:28:39.192142  [info] main dispatch thread id from application: 0100 (service-sample) is: 6
2025-01-23 15:28:39.192152  [error] Local UDS server endpoint initialization failed. Client 0100 Path: /var/vsomeip-100 Reason: Protocol not available
2025-01-23 15:28:39.192152  [info] shutdown thread id from application: 0100 (service-sample) is: 7
2025-01-23 15:28:39.192162  [info] OFFER(0100): [1234.5678:1.0] (true)
Successfully Registered Service!
2025-01-23 15:28:39.192162  [error] create_server_endpoint Server endpoint creation failed. Reason: Invalid argument Port: 30490 (0)
2025-01-23 15:28:39.192172  [error] Service Discovery endpoint could not be created. Please check your network configuration.
2025-01-23 15:28:39.192172  [error] Couldn't start service discovery
2025-01-23 15:28:39.192172  [info] Port configuration missing for [1234.5678]. Service is internal.
2025-01-23 15:28:39.192172  [info] rmi::start_ip_routing: clear pending_sd_offers_
2025-01-23 15:28:39.192182  [info] SOME/IP routing ready.
2025-01-23 15:28:39.192182  [info] create_routing_root: Routing root @ /var/vsomeip-0
2025-01-23 15:28:39.192182  [error] Local routing endpoint creation failed. Client ID: 0000: Protocol not available
2025-01-23 15:28:39.192192  [warning] Routing root creating (partially) failed. Please check your configuration.
2025-01-23 15:28:39.192192  [info] Watchdog is disabled!
2025-01-23 15:28:39.192202  [info] io thread id from application: 0100 (service-sample) is: 9
2025-01-23 15:28:39.192202  [info] io thread id from application: 0100 (service-sample) is: 5
2025-01-23 15:28:39.192202  [info] vSomeIP 3.5.4 | (default)

### Reproduction Steps

echo -e "[default]\nbinding=someip\n\n[logging]\nconsole = true\nfile = ./mylog.log\ndlt = true\nlevel = verbose" > /tmp/commonapi.ini

export VSOMEIP_CONFIGURATION=/mnt/usr/etc/

export VSOMEIP_CONFIGURATION=/mnt/usr/etc/vsomeip.json

export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/mnt/usr/lib

export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/mnt/tmp

cd /mnt/bin

COMMONAPI_CONFIG=/tmp/commonapi.ini ./BroadcastService

### Logs and Screenshots

[image]

AhMmedMahmoud, 2025-02-05:
update
vsomeip.json
{
"unicast" : "127.0.0.1",
"logging" :
{
"level" : "debug",
"console" : "true",
"file" : { "enable" : "false", "path" : "/var/log/vsomeip.log" },
"dlt" : "false"
},
"applications" :
[
{
"name" : "client-sample",
"id" : "0x1343"
},
{
"name" : "other-client-sample",
"id" : "0x1344"
},
{
"name" : "service-sample",
"id" : "0x1277"
}
],
"services" :
[
{
"service" : "0x1234",
"instance" : "0x5678",
"reliable" : { "port" : "30509", "enable-magic-cookies" : "false" },
"unreliable" : "31000"
},
{
"service" : "0x1235",
"instance" : "0x5678",
"reliable" : { "port" : "30506", "enable-magic-cookies" : false },
"unreliable" : "31000"
}
],
"routing" : "server-sample",
"service-discovery" :
{
"enable" : "false",
"multicast" : "224.244.224.245",
"port" : "30490",
"protocol" : "udp",
"initial_delay_min" : "10",
"initial_delay_max" : "100",
"repetitions_base_delay" : "200",
"repetitions_max" : "3",
"ttl" : "3",
"cyclic_offer_delay" : "2000",
"request_response_delay" : "1500"
}
}

duartenfonseca, 2025-04-01:
@AhMmedMahmoud can you make sure you have all the permissions on the /var/ folder? or if you should be using like /var/tmp/ as the VSOMEIP_BASE_PATH