COVESA/vsomeip #850: [BUG]: <title>Can't work cross a routed LANs.

AricWang-E2E, 2025-02-21:
### vSomeip Version

v3.1.37.1

### Boost Version

1.7.1

### Environment

linux  / android

### Describe the bug

 Host A(eth0) <--------->HostB(eth0, eth1)<----------->HostC(eth0), HostB act as a router(third layer routing and SMCRoute).  While vsomeip application run on HostA and HostC with different sub net. 
HostA sub net: 192.168.7.0/24 
HostC sub net: 192.168.4.0/24
vsomeip service multicast packets could be routed by HostB. But vsomeip client recevied multicast packets, but will not try to connect register service.

I have a question: If VSOMEIP was designed unicast only?  Only works in one sub net ? Thanks in advanced!

KareemLMR, 2025-02-21:
Hello @AricWang-E2E 
I think both the server and the client must be in the same subnet. There is a function in the vsomeip code that rejects packets if sent from a client from different subnet so full communication between server and client doesn't work.
I have debugged this issue before here: https://github.com/COVESA/vsomeip/issues/800
It has the same issue you faced I think. And the summary of the root cause you will find in the last few comments.

If you must use a router and make the server in different subnet from the client in your project, then you may try one of the solutions mentioned in the same issue link of commenting this condition (is_same_subnet()) here:

[image]

I think that will work with you. Try it and tell me what you found.

KareemLMR, 2025-02-25:
You are welcome!
I'm glad that it is working with you now.

AricWang-E2E, 2025-02-26:
@duartenfonseca  Why close this ticket? and without any explanation. Why vsomeip designed as only working  with the same LAN? If so what's the point of multicast?

duartenfonseca, 2025-02-26:
@AricWang-E2E I was under the impression that you had it working now. one application on multiple vlans is not possible, multiple applications on different vlans it is possible

duartenfonseca, 2025-03-24:
@AricWang-E2E @KareemLMR  if you want you can add this as a PR, but this is not a bug, this is simply not the use case that was had in mind when creating the lib.

closed 2025-03-24 by duartenfonseca

yanxiaochun1005, 2025-12-24:
time to live(ttl) was set by default was 1,how to fix it

KareemLMR, 2025-12-30:
It can be changed from the json file vsomeip.json