COVESA/vsomeip #852: [BUG]: <title>service-example is not working in VM

kjitender469, 2025-02-25:
### vSomeip Version

1f4364db98bcdb8431acfb3563f71eccf46040ae

### Boost Version

1.74

### Environment

Ubuntu 22.04 running inside VM

### Describe the bug

service-example.cpp is not working as expected. Not getting Received message with Client/Session [1001/0001] 00 01 02 03 04 05 06 07 08 09.

**Steps to reproduce: **
=========================
1. Install Ubuntu 22.04 in Oracle VM
2. Follow **https://github.com/COVESA/vsomeip/wiki/vsomeip-in-10-minutes**

**cmake logs:**
========================
vboxuser@Ubuntu-22:~/Desktop/someip_ws/vsomeip/build$ cmake ..
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Setting build type to 'RelWithDebInfo' as none was specified.
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.66") found components: system thread filesystem 
-- Using boost version: 107400
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
-- Systemd was not found, watchdog disabled!
-- Default configuration folder: /etc/vsomeip
-- Default configuration file: /etc/vsomeip.json
Predefined base path: /tmp
Predefined unicast address: 127.0.0.1
Predefined diagnosis address: 0x01
Predefined wait times for internal communication ports (TCP): 100 (max=10000)
-- Found Doxygen: /usr/bin/doxygen (found version "1.9.1") found components: doxygen dot 
dot found
CMake Warning at test/CMakeLists.txt:12 (find_package):
  By not providing "Findbenchmark.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "benchmark", but CMake did not find one.

  Could not find a package configuration file provided by "benchmark" with
  any of the following names:

    benchmarkConfig.cmake
    benchmark-config.cmake

  Add the installation prefix of "benchmark" to CMAKE_PREFIX_PATH or set
  "benchmark_DIR" to a directory containing one of the above files.  If
  "benchmark" provides a separate development package or SDK, be sure it has
  been installed.

-- GTEST_ROOT is not defined. For building the tests the variable
             GTEST_ROOT has to be defined. Tests can not be built.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/vboxuser/Desktop/someip_ws/vsomeip/build
vboxuser@Ubuntu-22:~/Desktop/someip_ws/vsomeip/build$ 

**service-example logs**
==========================
vboxuser@Ubuntu-22:~/Desktop/someip_ws/vsomeip/test_example$ LD_LIBRARY_PATH=../../install_folder/lib/:$PWD/build/ ./build/service-example
2025-02-25 17:00:38.027719  [info] Parsed vsomeip configuration in 0ms
2025-02-25 17:00:38.028286  [info] Configuration module loaded.
2025-02-25 17:00:38.028338  [info] Security disabled!
2025-02-25 17:00:38.028429  [info] Initializing vsomeip (3.5.4) application "World".
2025-02-25 17:00:38.032635  [info] Instantiating routing manager [Host].
2025-02-25 17:00:38.035003  [info] create_routing_root: Routing root @ /tmp/vsomeip-0
2025-02-25 17:00:38.036402  [info] Service Discovery enabled. Trying to load module.
2025-02-25 17:00:38.038559  [info] Service Discovery module loaded.
2025-02-25 17:00:38.039925  [info] Application(World, 0100) is initialized (11, 100).
2025-02-25 17:00:38.040250  [info] rmi::offer_service added service: 1234 to pending_sd_offers_.size = 1
2025-02-25 17:00:38.041334  [info] create_local_server: Listening @ /tmp/vsomeip-100
2025-02-25 17:00:38.041612  [info] OFFER(0100): [1234.5678:0.0] (true)
2025-02-25 17:00:38.041691  [info] Starting vsomeip application "World" (0100) using 2 threads I/O nice 0
2025-02-25 17:00:38.046158  [info] main dispatch thread id from application: 0100 (World) is: 73b427200640 TID: 7270
2025-02-25 17:00:38.047694  [info] Client [0100] routes unicast:127.0.0.1, netmask:255.255.255.0
2025-02-25 17:00:38.047866  [info] shutdown thread id from application: 0100 (World) is: 73b426800640 TID: 7271
2025-02-25 17:00:38.049870  [info] Watchdog is disabled!
2025-02-25 17:00:38.054966  [info] io thread id from application: 0100 (World) is: 73b42893ab80 TID: 7268
2025-02-25 17:00:38.055696  [info] vSomeIP 3.5.4 | (default)
2025-02-25 17:00:38.056072  [warning] Network interface "lo" state changed: up
2025-02-25 17:00:38.055414  [info] io thread id from application: 0100 (World) is: 73b425400640 TID: 7273
2025-02-25 17:00:47.590375  [info] Application/Client 0101 is registering.
2025-02-25 17:00:47.595863  [info] emb::find_or_create_local: create_client 101
2025-02-25 17:00:47.597263  [info] Client [100] is connecting to [101] at /tmp/vsomeip-101 endpoint > 0x73b40c000e70
2025-02-25 17:00:47.613142  [info] REGISTERED_ACK(0101)
2025-02-25 17:00:47.614622  [info] REGISTER EVENT(0101): [1234.5678.8778:eventtype=0:is_provided=false:reliable=255]
2025-02-25 17:00:47.636842  [info] REQUEST(0101): [1234.5678:255.4294967295]
2025-02-25 17:00:47.655674  [info] SUBSCRIBE(0101): [1234.5678.4465:ffff:0]
2025-02-25 17:00:48.062120  [info] vSomeIP 3.5.4 | (default)
2025-02-25 17:00:58.071673  [info] vSomeIP 3.5.4 | (default)

**client-example logs**
=========================
vboxuser@Ubuntu-22:~/Desktop/someip_ws/vsomeip/test_example$ LD_LIBRARY_PATH=../../install_folder/lib/:$PWD/build/ ./build/client-example
2025-02-25 17:00:47.528488  [info] Parsed vsomeip configuration in 0ms
2025-02-25 17:00:47.529366  [info] Configuration module loaded.
2025-02-25 17:00:47.529455  [info] Security disabled!
2025-02-25 17:00:47.529580  [info] Initializing vsomeip (3.5.4) application "Hello".
2025-02-25 17:00:47.529846  [info] Instantiating routing manager [Proxy].
2025-02-25 17:00:47.530277  [info] Client [ffff] is connecting to [0] at /tmp/vsomeip-0 endpoint > 0x5bf839525680
2025-02-25 17:00:47.530384  [info] Application(Hello, ffff) is initialized (11, 100).
2025-02-25 17:00:47.535946  [info] Starting vsomeip application "Hello" (ffff) using 2 threads I/O nice 0
2025-02-25 17:00:47.540710  [info] main dispatch thread id from application: ffff (Hello) is: 7e8cd0e00640 TID: 7277
2025-02-25 17:00:47.570610  [info] shutdown thread id from application: ffff (Hello) is: 7e8cd0400640 TID: 7278
CLIENT: Service [1234.5678] is NOT available.
2025-02-25 17:00:47.543131  [info] io thread id from application: ffff (Hello) is: 7e8cd251fb80 TID: 7275
2025-02-25 17:00:47.577653  [info] assign_client: (ffff:Hello)
2025-02-25 17:00:47.543509  [info] io thread id from application: ffff (Hello) is: 7e8ccfa00640 TID: 7279
2025-02-25 17:00:47.587055  [info] create_local_server: Listening @ /tmp/vsomeip-101
2025-02-25 17:00:47.588586  [info] Client 0101 (Hello) successfully connected to routing  ~> registering..
2025-02-25 17:00:47.588871  [info] Registering to routing manager @ vsomeip-0
2025-02-25 17:00:47.606407  [info] Application/Client 0101 (Hello) is registered.
2025-02-25 17:00:47.640576  [info] emb::find_or_create_local: create_client 100
2025-02-25 17:00:47.643928  [info] Client [101] is connecting to [100] at /tmp/vsomeip-100 endpoint > 0x7e8cc00016c0
2025-02-25 17:00:47.645900  [info] ON_AVAILABLE(0101): [1234.5678:0.0]
CLIENT: Service [1234.5678] is available.
2025-02-25 17:00:47.657001  [info] SUBSCRIBE ACK(0100): [1234.5678.4465.ffff]

kjitender469, 2025-02-27:
After running service and client multiple times at some point it works. I started client first then start and stop service for 4 5 times after that stopped client and again start client then it shows the expected output.

duartenfonseca, 2025-04-01:
@kjitender469  can you provide the code you are using? thanks