COVESA/vsomeip #857: [BUG]: The proxy subscription status is incorrect

yanxiaochun1005, 2025-02-28:
### vSomeip Version

V3.4.10 

### Boost Version

1.7.4

### Environment

Ubuntu 22.04

### Describe the bug

The proxy mode is used for communication. When the TTL is triggered, the process of canceling and subscriping is carried out. At this time, it is found that the subscription status of the routing management module on the host is inconsistent with that on the proxy side.

yanxiaochun1005, 2025-02-28:
My current modification policy is that the subscription message will be routed to the agent, without judging whether the current subscription is ok.
I wonder if there are any other problems with the above strategy.

continue1913, 2025-03-01:
[image]

duartenfonseca, 2025-04-01:
could you try and summarize the issue? it was a bit hard to understand.
what would happen if you changed ttl to 10 seconds?

yanxiaochun1005, 2025-12-24:
> [image] When system load is high, the sub from external app expired, and at the moment, host process sub，stop sub message, send these thress message to proxy, after sent, the proxy send back unsub ack, suback, unsub ack, and now proxy recored no remote sub, the host recored one remote sub and acked, there for the proxy wouldn't send event message to external app，until anyone of them restart。 We want to delete the code "clients_.erase(its_client);" in function "remote_subscription::update" to resolve this problem, is it ok ?

yes，The latest version of this issue has been resolved.