COVESA/vsomeip #858: [BUG]: received server's OFFER has a one second cycle, why client "lost connection to remote service" ?

FishPickBeans, 2025-02-28:
### vSomeip Version

v3.4.10

### Boost Version

1.84

### Environment

Linux

### Describe the bug

The server OFFER occurs once a second, and after someip_cient starts, it keeps printing 'service_dedpoint-disconnected: lost connection to remote service: [1005.0001]'. Why is that?

### Reproduction Steps

Difficult to reproduce（I have currently discovered one）：
1. Embedded Linux development board, after OTA upgrade.

### Expected behaviour

The someip client is connected normally and can send and receive messages, method/Event.

### Logs and Screenshots

9 2025/02/26 12:04:50.158927 48.9724 0 ECU1 SPND VSIP 2214 log info verbose 1 Using configuration file: "/oemapp/conf/someip/vsomeip_normaldata.json".
10 2025/02/26 12:04:50.166087 48.9795 1 ECU1 SPND VSIP 2214 log info verbose 1 Parsed vsomeip configuration in 3ms
11 2025/02/26 12:04:50.166812 48.9803 2 ECU1 SPND VSIP 2214 log info verbose 1 Configuration module loaded.
12 2025/02/26 12:04:50.192571 49.0060 3 ECU1 SPND VSIP 2214 log info verbose 1 Security disabled!
13 2025/02/26 12:04:50.192744 49.0062 4 ECU1 SPND VSIP 2214 log info verbose 1 Initializing vsomeip (3.4.10) application "wl_someip_normaldata".
14 2025/02/26 12:04:50.193056 49.0065 5 ECU1 SPND VSIP 2214 log info verbose 1 Instantiating routing manager [Host].
15 2025/02/26 12:04:50.193904 49.0073 6 ECU1 SPND VSIP 2214 log info verbose 1 create_routing_root: Routing root @ /tmp/wl_someip_normaldata-0
16 2025/02/26 12:04:50.194348 49.0078 7 ECU1 SPND VSIP 2214 log info verbose 1 Service Discovery enabled. Trying to load module.
17 2025/02/26 12:04:50.195973 49.0094 8 ECU1 SPND VSIP 2214 log info verbose 1 Service Discovery module loaded.
18 2025/02/26 12:04:50.206894 49.0203 9 ECU1 SPND VSIP 2214 log info verbose 1 vsomeip tracing not enabled. . vsomeip service discovery tracing not enabled.
19 2025/02/26 12:04:50.207053 49.0205 10 ECU1 SPND VSIP 2214 log info verbose 1 Application(wl_someip_normaldata, 1f02) is initialized (11, 100).
20 2025/02/26 12:04:50.207779 49.0212 11 ECU1 SPND VSIP 2214 log info verbose 1 REGISTER EVENT(1f02): [1005.0001.8001:is_provider=false]
21 2025/02/26 12:04:50.207977 49.0214 12 ECU1 SPND VSIP 2214 log info verbose 1 REQUEST(1f02): [1005.0001:1.4294967295]
22 2025/02/26 12:04:50.208476 49.0219 13 ECU1 SPND VSIP 2214 log info verbose 1 create_local_server: Listening @ /tmp/wl_someip_normaldata-1f02
24 2025/02/26 12:04:50.210297 49.0237 14 ECU1 SPND VSIP 2214 log info verbose 1 Starting vsomeip application "wl_someip_normaldata" (1f02) using 2 threads I/O nice 255
25 2025/02/26 12:04:50.210807 49.0242 15 ECU1 SPND VSIP 2214 log info verbose 1 Client [1f02] routes unicast:172.16.59.10, netmask:255.255.255.0
26 2025/02/26 12:04:50.211566 49.0250 16 ECU1 SPND VSIP 2214 log info verbose 1 Watchdog is disabled!
27 2025/02/26 12:04:50.212022 49.0255 17 ECU1 SPND VSIP 2214 log info verbose 1 io thread id from application: 1f02 (wl_someip_normaldata) is: b3fcd3c0 TID: 2483
29 2025/02/26 12:04:50.217161 49.0306 19 ECU1 SPND VSIP 2214 log info verbose 1 vSomeIP 3.4.10 | (default)
31 2025/02/26 12:04:50.217703 49.0312 21 ECU1 SPND VSIP 2214 log info verbose 1 shutdown thread id from application: 1f02 (wl_someip_normaldata) is: b2dfe3c0 TID: 2485
32 2025/02/26 12:04:50.217927 49.0314 22 ECU1 SPND VSIP 2214 log info verbose 1 main dispatch thread id from application: 1f02 (wl_someip_normaldata) is: b35ff3c0 TID: 2484
34 2025/02/26 12:04:50.219734 49.0332 23 ECU1 SPND VSIP 2214 log info verbose 1 io thread id from application: 1f02 (wl_someip_normaldata) is: b1dfc3c0 TID: 2487
42 2025/02/26 12:04:52.584935 51.3983 31 ECU1 SPND VSIP 2214 log info verbose 1 Network interface "eth0.59" state changed: down
45 2025/02/26 12:04:52.633388 51.4467 34 ECU1 SPND VSIP 2214 log info verbose 1 Network interface "eth0.59" state changed: down
47 2025/02/26 12:04:52.781031 51.5943 36 ECU1 SPND VSIP 2214 log info verbose 1 Network interface "eth0.59" state changed: up
193 2025/02/26 12:05:29.274361 59.0309 182 ECU1 SPND VSIP 2214 log info verbose 1 vSomeIP 3.4.10 | (default)
201 2025/02/26 12:05:29.584863 59.3404 190 ECU1 SPND VSIP 2214 log info verbose 1 Route "239.0.0.255/32 if: eth0.59 gw: n/a" state changed: up
202 2025/02/26 12:05:29.584919 59.3407 191 ECU1 SPND VSIP 2214 log info verbose 1 udp_server_endpoint_impl: SO_RCVBUFFORCE successful.
203 2025/02/26 12:05:29.584953 59.3408 192 ECU1 SPND VSIP 2214 log info verbose 1 udp_server_endpoint_impl: SO_RCVBUF is: 1703936 (1703936) local port:30490
204 2025/02/26 12:05:29.584984 59.3411 193 ECU1 SPND VSIP 2214 log info verbose 1 SOME/IP routing ready.
205 2025/02/26 12:05:29.585014 59.3414 194 ECU1 SPND VSIP 2214 log info verbose 1 udp_server_endpoint_impl<multicast>: SO_RCVBUFFORCE: successful.
206 2025/02/26 12:05:29.585044 59.3415 195 ECU1 SPND VSIP 2214 log info verbose 1 udp_server_endpoint_impl<multicast>: SO_RCVBUF is: 1703936 (1703936) local port:30490
207 2025/02/26 12:05:29.635348 59.3913 196 ECU1 SPND VSIP 2214 log info verbose 1 configuration_impl:find_specific_port #1: service: 1005 instance: 1 reliable: 1 return specific port: 30501
208 2025/02/26 12:05:29.635411 59.3915 197 ECU1 SPND VSIP 2214 log info verbose 1 endpoint_manager_impl::create_remote_client: 172.16.59.3:30500 reliable: 1 using local port: 30501
209 2025/02/26 12:05:29.653079 59.4029 198 ECU1 SPND VSIP 2214 log info verbose 1 SUBSCRIBE(1f02): [1005.0001.0003:8001:1]
217 2025/02/26 12:05:30.312480 60.0677 200 ECU1 SPND VSIP 2214 log warn verbose 1 tce::restart: local: 172.16.59.10:30501 remote: 172.16.59.3:30500
218 2025/02/26 12:05:30.411812 60.1683 201 ECU1 SPND VSIP 2214 log warn verbose 1 service_endpoint_disconnected: lost connection to remote service: [1005.0001]
222 2025/02/26 12:05:30.512376 60.2688 202 ECU1 SPND VSIP 2214 log warn verbose 1 service_endpoint_disconnected: lost connection to remote service: [1005.0001]
223 2025/02/26 12:05:30.713389 60.4697 203 ECU1 SPND VSIP 2214 log warn verbose 1 service_endpoint_disconnected: lost connection to remote service: [1005.0001]
224 2025/02/26 12:05:31.113916 60.8703 204 ECU1 SPND VSIP 2214 log warn verbose 1 service_endpoint_disconnected: lost connection to remote service: [1005.0001]
225 2025/02/26 12:05:31.308755 61.0651 205 ECU1 SPND VSIP 2214 log warn verbose 1 tce::restart: dropping message: remote:172.16.59.3:30500 (1f02): [1005.0001.000f] size: 39
226 2025/02/26 12:05:31.308816 61.0652 206 ECU1 SPND VSIP 2214 log warn verbose 1 tce::restart: local: remote: 172.16.59.3:30500
228 2025/02/26 12:05:32.909382 62.6658 208 ECU1 SPND VSIP 2214 log warn verbose 1 service_endpoint_disconnected: lost connection to remote service: [1005.0001]
[... 100 lines cut ...]
538 2025/02/26 12:45:49.409072 2479.2077 3 ECU1 SPND VSIP 2214 log info verbose 1 vSomeIP 3.4.10 | (default)
539 2025/02/26 12:45:59.409285 2489.2079 4 ECU1 SPND VSIP 2214 log info verbose 1 vSomeIP 3.4.10 | (default)
540 2025/02/26 12:46:09.409611 2499.2081 5 ECU1 SPND VSIP 2214 log info verbose 1 vSomeIP 3.4.10 | (default)
541 2025/02/26 12:46:09.409669 2499.2083 6 ECU1 SPND VSIP 2214 log info verbose 1 ECQ: [172.16.59.3:30500(tcp):116509]
542 2025/02/26 12:46:19.410087 2509.2086 7 ECU1 SPND VSIP 2214 log info verbose 1 vSomeIP 3.4.10 | (default)
543 2025/02/26 12:46:29.410311 2519.2089 8 ECU1 SPND VSIP 2214 log info verbose 1 vSomeIP 3.4.10 | (default)
544 2025/02/26 12:46:39.410490 2529.2091 9 ECU1 SPND VSIP 2214 log info verbose 1 vSomeIP 3.4.10 | (default)
545 2025/02/26 12:46:49.410661 2539.2093 10 ECU1 SPND VSIP 2214 log info verbose 1 vSomeIP 3.4.10 | (default)
546 2025/02/26 12:46:59.411996 2549.2098 11 ECU1 SPND VSIP 2214 log info verbose 1 vSomeIP 3.4.10 | (default)
547 2025/02/26 12:47:09.411672 2559.2102 12 ECU1 SPND VSIP 2214 log info verbose 1 vSomeIP 3.4.10 | (default)
548 2025/02/26 12:47:09.411750 2559.2103 13 ECU1 SPND VSIP 2214 log info verbose 1 ECQ: [172.16.59.3:30500(tcp):119449]
1774 2025/02/26 15:42:20.072586 13069.8712 215 ECU1 SPND VSIP 2214 log info verbose 1 vSomeIP 3.4.10 | (default)
1775 2025/02/26 15:42:30.154991 13079.8715 216 ECU1 SPND VSIP 2214 log info verbose 1 vSomeIP 3.4.10 | (default)
1776 2025/02/26 15:42:40.073629 13089.8720 217 ECU1 SPND VSIP 2214 log info verbose 1 vSomeIP 3.4.10 | (default)
1777 2025/02/26 15:42:50.074016 13099.8724 218 ECU1 SPND VSIP 2214 log info verbose 1 vSomeIP 3.4.10 | (default)
1778 2025/02/26 15:43:00.074105 13109.8727 219 ECU1 SPND VSIP 2214 log info verbose 1 vSomeIP 3.4.10 | (default)
[... cut ...]
1783 2025/02/26 15:43:40.075380 13149.8740 224 ECU1 SPND VSIP 2214 log info verbose 1 vSomeIP 3.4.10 | (default)
1784 2025/02/26 15:43:50.075576 13159.8742 225 ECU1 SPND VSIP 2214 log info verbose 1 vSomeIP 3.4.10 | (default)
1785 2025/02/26 15:44:01.091292 13169.8746 226 ECU1 SPND VSIP 2214 log info verbose 1 vSomeIP 3.4.10 | (default)
1786 2025/02/26 15:44:10.076304 13179.8748 227 ECU1 SPND VSIP 2214 log info verbose 1 vSomeIP 3.4.10 | (default)
1787 2025/02/26 15:44:10.076361 13179.8749 228 ECU1 SPND VSIP 2214 log info verbose 1 ECQ: [172.16.59.3:30500(tcp):639878]

FishPickBeans, 2025-02-28:
txt log:
[log1.txt](https://github.com/user-attachments/files/19025054/log1.txt)

tcpdump:
[tcpdump_20250226120531_00.pcap.gz](https://github.com/user-attachments/files/19025147/tcpdump_20250226120531_00.pcap.gz)

duartenfonseca, 2025-04-01:
@FishPickBeans 
are we seeing client side logs, service side logs, or both?

try again using 30491 as client port and 30501 as service.
apart to this you can also try using 3.5.5, because we have done some changes on those parts of the code

FishPickBeans, 2025-04-27:
thank you, we had use 3.5.5 testing!

closed 2025-04-27 by FishPickBeans