COVESA/vsomeip #881: [BUG]: <title>hello_world_services 崩溃

zhangp11, 2025-04-08:
### vSomeip Version

v3.5.5

### Boost Version

1.8.5

### Environment

window11 vs2022

### Describe the bug

[image]
我在使用以下配置文件时hello_world_services崩溃，发生了数组越界。我使用liunx系统可以正常运行

[helloworld-local.json](https://github.com/user-attachments/files/19643277/helloworld-local.json)

goncaloalmeida, 2025-04-08:
Could you please put your question in English? The Hello World example was disabled for Windows, right @duartenfonseca?

https://github.com/COVESA/vsomeip/blob/master/CMakeLists.txt#L661

zhangp11, 2025-04-09:
Not just helloworld, in “vsomeip/implementation/protocol/src/config_command.cpp”

“value.assign(&_buffer[read_position], &_buffer[read_position + size]);” I printed the value of the helloworld run in liunx
buffer size:40 read_postion:13 size 8
buffer size:40 read_postion:25 size 15 This place array was sent out of bounds, and liunx reported no error
I modified it like this, “value.assign(&_buffer[read_position], &_buffer[std::min(read_position + size, _buffer.size())])”; But not sure if there is an error

zhangp11, 2025-04-09:
[image]
This error has nothing to do with hello_world, it's vsomeip.dll an error occurred, I used request-sample, it still crashes, I use gcc a little less, I don't know if there will be an error after the memory is out of bounds.。However, the above hello-world has a memory out-of-bounds under liunx, but it does not crash