COVESA/vsomeip #905: suddenly stops sending multicast packets from the sever

sureshsakhamuri, 2025-05-30:
### vSomeip Version

v3.1.20.3

### Boost Version

1.70

### Environment

linux5.2

### Describe the bug

Hi,
     Suddenly stop sending any packet on port 30490 from the server. Couls you please let me know is there any know issue in 3.1.20.3?
logs:
======
[info]	service_discovery_implon_main_phase_timer_expired:3120												
[info]	serialize_and_send:2515												
[info]	Exit	send_to210											
[info]	find_eventgroup:1089												
[info]	Exit	send_to210											
[info]	send:997_is_announcing	%d1											
[info]	Entry	send_to207											
[info]	Entry	send_to207											
[info]	insert_offer_entries:710												
[info]	is_suspended_	%d0											
[info]	add_entry_data:111												
[info]	_ignore_phase	%d0											
[info]	find_eventgroup:2010												
[info]	before	offer	service										
[info]	find_eventgroup:2010												
[info]	add_entry_data												
[info]	Entry	send_to207											
[info]	Entry	send_to207											
[info]	add_entry_data:3713												
[info]	serialize_and_send:2515												
[info]	add_entry_data:111												
[info]	Entry	on_remote_subscribe2759											
[info]	send:2499												
[info]	Entry	on_multicast_received537											
[info]	Routing	.	Endpoint	for	service	(0000):	[ffff.0000.8100]						
[info]	before	put_serializer	send:856										
[info]	Exit	send_to210											
[info]	Exit	send_to210											
[info]	Entry	on_message1146											
[info]	Entry	on_remote_subscribe2759											
[info]	Entry	on_message1146											
[info]	send_intern:245												
[info]	Exit	send_to210											
[info]	put_serializer:1311Queue	size:	%d0										
[info]	send:745												
[info]	Exit	send_to210											
[info]	find_eventgroup:1089												
[info]	send_intern:245												
[info]	send_intern:245												
[error]	process_eventgroupentry:	Received	a	SubscribeEventGroup	entry	for	unknown	eventgroup	100d.0002.000c]	session:	009a,	ttl:	4
[info]	send_intern:245												
[info]	add_entry_data:111												
[info]	Entry	get_configured_times_from_endpoint272											
[info]	add_entry_data:111												
[info]	add_entry_data:111
Thanks,
Suresh Sakhamuri.

### Expected behaviour

Hi,
   suddenly stops sending multicast packets from the sever and any packet on port 30490. Could you please is there any know issue in 3.1.20.3
logs:
=====
[info]	service_discovery_implon_main_phase_timer_expired:3120												
[info]	serialize_and_send:2515												
[info]	Exit	send_to210											
[info]	find_eventgroup:1089												
[info]	Exit	send_to210											
[info]	send:997_is_announcing	%d1											
[info]	Entry	send_to207											
[info]	Entry	send_to207											
[info]	insert_offer_entries:710												
[info]	is_suspended_	%d0											
[info]	add_entry_data:111												
[info]	_ignore_phase	%d0											
[info]	find_eventgroup:2010												
[info]	before	offer	service										
[info]	find_eventgroup:2010												
[info]	add_entry_data												
[info]	Entry	send_to207											
[info]	Entry	send_to207											
[info]	add_entry_data:3713												
[info]	serialize_and_send:2515												
[info]	add_entry_data:111												
[info]	Entry	on_remote_subscribe2759											
[info]	send:2499												
[info]	Entry	on_multicast_received537											
[info]	Routing	.	Endpoint	for	service	(0000):	[ffff.0000.8100]						
[info]	before	put_serializer	send:856										
[info]	Exit	send_to210											
[info]	Exit	send_to210											
[info]	Entry	on_message1146											
[info]	Entry	on_remote_subscribe2759											
[info]	Entry	on_message1146											
[info]	send_intern:245												
[info]	Exit	send_to210											
[info]	put_serializer:1311Queue	size:	%d0										
[info]	send:745												
[info]	Exit	send_to210											
[info]	find_eventgroup:1089												
[info]	send_intern:245												
[info]	send_intern:245												
[error]	process_eventgroupentry:	Received	a	SubscribeEventGroup	entry	for	unknown	eventgroup	100d.0002.000c]	session:	009a,	ttl:	4
[info]	send_intern:245												
[info]	add_entry_data:111												
[info]	Entry	get_configured_times_from_endpoint272											
[info]	add_entry_data:111												
[info]	add_entry_data:111
Thanks,
Suresh Sakhamuri.

duartenfonseca, 2025-06-02:
hi @sureshsakhamuri we don't support that version, can you confirm if this happens with more recent versions?
Thanks

raghavendrard, 2025-06-03:
> hi [@sureshsakhamuri](https://github.com/sureshsakhamuri) we don't support that version, can you confirm if this happens with more recent versions? Thanks

Hi @duartenfonseca ..
   Thanks for the reply. Due to final phase of the project, we cannot update to new version. 
Can you share some insight, if this is a known issue, patch available with fix etc?

raghavendrard, 2025-06-05:
Hi @duartenfonseca ..
   We have implemented fall back mechanism to handle this case. 

Kindly review and confirm ! Your review is required for our customer to give a go ahead. So please support us !
  
[image]

Patch: 

[Patch_vSomeIP_FallBackMechanism.patch](https://github.com/user-attachments/files/20611575/Patch_vSomeIP_FallBackMechanism.patch)

cc - @sureshsakhamuri 

Regards
Raghavendra