COVESA/vsomeip #913: [BUG]: Why was the fix in commit 338d6090 removed in recent changes?

jinchangrong-jcr, 2025-06-17:
### vSomeip Version

3.5.0

### Boost Version

1.74

### Environment

linux

### Describe the bug

Body:

I noticed that the changes introduced in commit [338d6090](https://github.com/COVESA/vsomeip/commit/338d6090113d514f28757e30f417bb6ee643f0f3) (which fixed [briefly describe the problem this commit addressed]) appear to have been removed in the latest code updates.
vsomeip 3.1.37.1:
[image]

 I believe this change was a correct and necessary fix. 

Could the maintainers please clarify:
What was the reason for reverting this change?
Is there an alternative solution being considered?
Would it be possible to re-introduce this fix if the original problem still exists?
I'm happy to help test or verify this behavior if needed. Thank you for your time and consideration!

duartenfonseca, 2025-06-17:
hey @jinchangrong-jcr i think this commit was erased by mistake, i'm sorry for that. 
For the interest of testing correctly, and ensuring it would make sense, can you give me a reproduction of the original error? 

thanks

jinchangrong-jcr, 2025-06-18:
Thanks for your reply!
Actually, in our production environment using vsomeip 3.1.20.3 with TCP request/response mode. After triggering messages between stop_offer_service and offer_service, the system enters an unrecoverable state where subsequent service offerings fail permanently.
Key Symptoms & Logs:

Initial Error Trigger:
[warn]: Didn't find connection: 10.220.227.195:64689 dropping outstanding messages (1)
This warning indicates failed message delivery due to missing TCP connection.

Subsequent Failures:
insert_offer_command returns false during rmi::offer_service:
rmi::offer_service (1342): [1020.3040:0.0] (true) not offering service, because insert_offer_command returned false!
stop_offer_service fails to insert STOP-OFFER commands:
rmi::stop_offer_service (1342): [1020.3040:0.0] (true) STOP-OFFER NOT INSERTED!
Message Delivery Failure:
find_local_client fails in routing_manager_impl::on_message:
cpp
if (utility::is_request(_data[VSOMEIP_MESSAGE_TYPE_POS])) {
    its_client = find_local_client(_service, _instance);
}
This forces the system to send requests to proxies instead of local services, resulting in:
Routing info for remote service could not be found! (1343): [1020.3040.5060] 0008
Root Cause Analysis:
The issue stems from incorrect handling of prepare_stop_handlers_ in server_endpoint_impl<Protocol>::prepare_stop:

Sequence Violation:
queue_train(...) is called first, which triggers send_queued(...).
If the connection is invalid at this point, send_queued finds prepare_stop_handlers_ empty and skips processing.
Then prepare_stop_handlers_[_service] = _handler is executed, but the handlers are never consumed.

The queued_train = true flag causes the system to assume handlers are queued, but the actual handler insertion happens after the queue operation.
This results in unprocessed handlers remaining in prepare_stop_handlers_, blocking future service offers.
Resolution & Workarounds:
The issue was resolved in [commit 60c83e0](https://github.com/GENIVI/vsomeip/commit/60c83e0b4beee7b8725a45112a83038640981a21) by reordering the prepare_stop method logic.
This ensures handlers are inserted before queue_train operations.

During the process of problem analysis, we found that:
vsomeip 3.1.37.1: Includes partial fixes for connection error handling in tcp_server_endpoint_impl::send_queued(),
void tcp_server_endpoint_impl::send_queued(const queue_iterator_type _queue_iterator) {
    connection::ptr its_connection = nullptr;
    {
...
        if (connection_iterator != connections_.end()) {
            its_connection = connection_iterator->second;
        } else {
            VSOMEIP_INFO << "Didn't find connection: "
                    << _queue_iterator->first.address().to_string() << ":" << std::dec
                    << static_cast<std::uint16_t>(_queue_iterator->first.port())
                    << " dropping outstanding messages (" << std::dec
                    << _queue_iterator->second.second.size() << ").";
            ...
                for (auto its_service : its_services) {
                    auto found_cbk = prepare_stop_handlers_.find(its_service);
                    VSOMEIP_INFO << "prepare_stop_handlers_ size:" << prepare_stop_handlers_.size();
                    if (found_cbk != prepare_stop_handlers_.end()) {
                        VSOMEIP_INFO << "Calling prepare stop handler "
                                << "for service: 0x"
                                << std::hex << std::setw(4) << std::setfill('0')
                                << its_service;
                        auto handler = found_cbk->second;
                        auto ptr = this->shared_from_this();
                        service_.post([ptr, handler, its_service](){
                                handler(ptr, its_service);
                        });
                        prepare_stop_handlers_.erase(found_cbk);
                    }
                }
            }
...
        }
    }
    if (its_connection) {
        its_connection->send_queued(_queue_iterator);
    }
}
We also found commit [338d6090](https://github.com/COVESA/vsomeip/commit/338d6090113d514f28757e30f417bb6ee643f0f3)

However, this commit is missing in later versions, potentially affecting scenarios with multiple services sharing the same queues_.
In our specific scenario, we only have a single service ID and instance ID, so commit 338d6090 does not affect us. However, I wonder about extreme cases where prepare_stop_handlers_ caches handlers for two service IDs (e.g., both service1 and service2 handlers exist simultaneously), but queues_ contains messages only for service1.
This would indicate that service2's messages have already been sent and its handler should be processed. However, due to the absence of commit 338d6090, the send_cbk might skip processing service2's handler, potentially triggering issues.
Thanks！

closed 2025-09-30 by jinchangrong-jcr