COVESA/vsomeip #925: [BUG]: When send SIGINT / SIGTERM to the  linux proccess creating the application it just hangs and doesn't Terminate

laurastroe, 2025-07-09:
### vSomeip Version

3.3.0

### Boost Version

1.74.0

### Environment

Ubunt 22.04, QNX

### Describe the bug

When compiled with -DENABLE_SIGNAL_HANDLING=1 the library is causing the linux process not  to Terminate when SIGTERm or SIGINT are sent.

### Reproduction Steps

compile libvsomeip3 with -DENABLE_SIGNAL_HANDLING=1  . 
Create at least two vsomeip application and start them from same thread (not main thread of the process )
Send SIGTERM/SIGINT 

Process is not terminated

### Expected behaviour

Process is terminated

When compiled with -DENABLE_SIGNAL_HANDLING=0 the process is behaving as expected and process is terminated 

### Logs and Screenshots

^C2025-07-08 14:01:54.474642 [info] Stopping vsomeip application "EVENT_MANAGER_OBA_VLAN110" (0102).
2025-07-08 14:01:54.474784 [info] Stopping vsomeip application "EVENT_MANAGER_OBA_VLAN10" (0101).
2025-07-08 14:01:54.474935 [info] Stopping vsomeip application "EVENT_MANAGER_OBA_VLAN14" (0100).
2025-07-08 14:01:54.479312 [info] Application/Client 0102 is deregistering.
2025-07-08 14:01:54.480002 [info] Application/Client 0101 is deregistering.
2025-07-08 14:01:54.481152 [info] receive_cbk local_uds_client_endpoint: connection_reset/EOF/bad_descriptor
2025-07-08 14:01:54.481400 [info] receive_cbk local_uds_client_endpoint: connection_reset/EOF/bad_descriptor
2025-07-08 14:01:54.481847 [info] Application/Client 101: Reconnecting to routing manager.
2025-07-08 14:01:54.482665 [info] Application/Client 102: Reconnecting to routing manager.
2025-07-08 14:01:54.490885 [info] Client [100] is closing connection to [102]
2025-07-08 14:01:54.491485 [info] RELEASE(0102): [00a0.0001]
2025-07-08 14:01:54.492540 [info] Client [100] is closing connection to [101]
2025-07-08 14:01:54.492972 [info] RELEASE(0101): [0016.0001]
2025-07-08 14:01:54.496316 [info] Exiting vsomeip application...
[main process] SomeIp::send_request
2025-07-08 14:01:54.515355 [error] Routing info for remote service could not be found! (0100): [0066.0001.0005] 0001
[main process]SomeIp::release_event
[main process] SomeIp::release_event
[main process]  SomeIp::send_request

^C^Z
[1]+  Stopped

duartenfonseca, 2025-07-11:
Hi @laurastroe does this also happen with 3.5.6? we don't support 3.3.0 anymore

laurastroe, 2025-07-11:
> Hi [@laurastroe](https://github.com/laurastroe) does this also happen with 3.5.6? we don't support 3.3.0 anymore

 , I see the same behaviour with 3.5.5 

2025-07-11 13:04:29.707104  [info] REQUEST(0102): [00a0.0001:1.4294967295]
^C2025-07-11 13:04:34.450969  [info] Stopping vsomeip application "APP0" (0101).
2025-07-11 13:04:34.450972  [info] Stopping vsomeip application "APP1" (0102).
2025-07-11 13:04:34.450977  [info] Stopping vsomeip application "APP2" (0100).
2025-07-11 13:04:34.451436  [info] Application/Client 0101 is deregistering.
2025-07-11 13:04:34.451506  [info] Application/Client 0102 is deregistering.
2025-07-11 13:04:34.451566  [info] emb::remove_local: client 101
2025-07-11 13:04:34.451632  [info] Application/Client 0101 (EVENT_MANAGER_OBA_VLAN10) is deregistered.
2025-07-11 13:04:34.451670  [info] cei::shutdown_and_close_socket_unlocked: not recreating socket  endpoint > 0x7e7d38000e50 socket state > 0
2025-07-11 13:04:34.451641  [info] local_uds_client_endpoint_impl::receive_cbk Error: Operation canceled
2025-07-11 13:04:34.451711  [info] Client [100] is closing connection to [101] endpoint > 0x7e7d38000e50
2025-07-11 13:04:34.451840  [info] RELEASE(0101): [0016.0001]
2025-07-11 13:04:34.451845  [info] local_uds_client_endpoint_impl::receive_cbk Error: Operation canceled
2025-07-11 13:04:34.451852  [info] cei::shutdown_and_close_socket_unlocked: not recreating socket  endpoint > 0x7e7d44011490 socket state > 0
2025-07-11 13:04:34.452212  [info] local_uds_client_endpoint_impl::receive_cbk Error: End of file
2025-07-11 13:04:34.452266  [info] rmc::handle_client_error: Client 0x0102 handles a client error 0x0000 with host, will reconnect
2025-07-11 13:04:34.452318  [info] emb::remove_local: client 100
2025-07-11 13:04:34.452345  [info] Application/Client 0102: Reconnecting to routing manager.
2025-07-11 13:04:34.452322  [info] emb::remove_local: client 102
2025-07-11 13:04:34.452457  [warning] cei::shutdown_and_close_socket_unlocked: socket has been reset  endpoint > 0x7e7d440176c0 socket state > 0
2025-07-11 13:04:34.452455  [info] local_uds_client_endpoint_impl::receive_cbk Error: Operation canceled
2025-07-11 13:04:34.452533  [warning] cei::shutdown_and_close_socket_unlocked: socket was not open  endpoint > 0x7e7d440176c0 socket state > 1
2025-07-11 13:04:34.452585  [info] cei::shutdown_and_close_socket_unlocked: not recreating socket  endpoint > 0x7e7d440176c0 socket state > 0
2025-07-11 13:04:34.452472  [info] cei::shutdown_and_close_socket_unlocked: not recreating socket  endpoint > 0x7e7d380024c0 socket state > 0
2025-07-11 13:04:34.452649  [info] Client [100] is closing connection to [102] endpoint > 0x7e7d380024c0
2025-07-11 13:04:34.452686  [info] RELEASE(0102): [00a0.0001]
^Z
[2]+  Stopped                 ./

I can check with 3.5.6 if you think might fix

jinchangrong-jcr, 2025-08-06:
hi @laurastroe , the reason is vsomeip signal handler only call stop not raise signal to system for its default action (terminate process)

[image]
in old version of vsomeip, vsomeip will call exit() on application_impl::start() after catch signal 2 or 15, so you can see the process terminate.
But, I don't think it's reasonable to call the exit() interface to terminate the process. The reason is that exit() will cause global variable destruction, which may trigger use-after-free issues and lead to crashes. If enabling VSOMEIP_ENABLE_SIGNAL_HANDLING in vsomeip is absolutely necessary, I suggest modifying its signal handler function to call stop() first, then raise the corresponding signal number to allow the system's default signal handling process to proceed.
cc @duartenfonseca

closed 2026-04-21 by fcmonteiro